fellesverkstedet/fabricatable-machines #9: Chamfer rail motor plate has too short slots

Jaknil, 2017-12-11:
The current chamfer rail motor plate has too short slots for the pinion to be moved completely out of the rack teeth when tuning the glide blocks. They need to be extended at least 3.6 mm on the inside for it to be possible. See pic.

[image]

JensDyvik, 2017-12-12:
Thank you for good feedback and well explaining screenshot. The missing feature has been added to the chamfer rail generator and the example files have been updated here https://github.com/fellesverkstedet/fabricatable-machines/tree/master/chamferrail

The downside of the is that the adjustment holes become very long. Especially since it is nice to to have the glidblock symmetric for ease off assembly. I increased the default wall thickness outside the holes from 4mm to 6mm for keeping a robust motor plate, also in weak materials

JensDyvik, 2017-12-12:
The missing feature has been added to the chamfer rail generator and the example files have been updated here https://github.com/fellesverkstedet/fabricatable-machines/tree/master/chamferrail

closed 2017-12-12 by JensDyvik