fellesverkstedet/fabricatable-machines #14: Chamfer rail has too high friction under heavy load

JensDyvik, 2018-06-08:
It seems that the chamfer rail axis system is not suitable for the heavy load of a heavy duty large format milling machine. This was discovered in the use of  [Humphrey2](https://github.com/fellesverkstedet/fabricatable-machines/tree/master/humphrey-large-format-cnc). Friction is too high when the plain bearings ride on the[ chamfer rail](https://github.com/fellesverkstedet/fabricatable-machines/tree/master/chamferrail). Aluminum oxide builds up on the plastic glide block surfaces, which leads to increasing friction with use.

Humphrey1 was a has a little lighter gantry than humphrey2. This could explain why humphrey1  has less problems with high friction. Total weight of gantry Humphrey1 = about 32kg. Total weight of gantry Humphrey2 = about 48kg.

Possible approaches to solve the friction challenge:

1. Find a way to coat the aluminum glide surfaces (UHMW tape, lacquer etc.)
2. Develop a roller bearing system that can be retrofitted to the machine ([relevant development by Jake Read at CBA](https://gitlab.cba.mit.edu/jakeread/rtcmill))
3. Use readymade linear rails from Aliexpress until we have a fabricatable axis suitable for heavy loads. Example of a 4000mm ready made [here](https://www.aliexpress.com/item/NEW-HIWIN-linear-guide-HGW25HA-CNC-rail-block-cnc-parts-for-HGR25-linear-guide-rails-block/1989551037.html?spm=2114.search0104.3.7.53af21a5xACF6P&ws_ab_test=searchweb0_0,searchweb201602_3_5722916_10152_10151_10065_10344_5722816_10068_10342_10343_10340_10341_5722616_10696_10084_10083_10618_10307_5722716_10059_100031_10103_10624_5722516_10623_10622_10621_10620,searchweb201603_2,ppcSwitch_5&algo_expid=c8b70173-733a-401b-b646-f562ce7e4bc7-1&algo_pvid=c8b70173-733a-401b-b646-f562ce7e4bc7&priceBeautifyAB=0) and [here](https://www.aliexpress.com/item/4pcs-lot-Cnc-Linear-Shaft-Chrome-OD-8mm-L-600mm-WCS-Round-Steel-Rod-Bar-Cylinder/32648144933.html?spm=2114.search0104.3.1.53af21a5xACF6P&ws_ab_test=searchweb0_0,searchweb201602_3_5722916_10152_10151_10065_10344_5722816_10068_10342_10343_10340_10341_5722616_10696_10084_10083_10618_10307_5722716_10059_100031_10103_10624_5722516_10623_10622_10621_10620,searchweb201603_2,ppcSwitch_5&algo_expid=c8b70173-733a-401b-b646-f562ce7e4bc7-0&algo_pvid=c8b70173-733a-401b-b646-f562ce7e4bc7&priceBeautifyAB=0).

What is a the best solution for improving Humphrey2? And what is the best solution for future fabricatable machines that need to handle high loads with high precision?

uomou, 2018-06-08:
You might give it a try to use PTFE instead of aertaliet.
I can send you some if you wish. I have a hole roll, enough for many humphreys to come

Met vriendelijke groet,

Alex

fabguru | Alex Schaub
“specialism within the global maker movement”
http://fabguru.com <http://fabguru.com/>

follow:
https://twitter.com/fabguru2015 <https://twitter.com/fabguru2015>
https://www.facebook.com/fabguru2015 <https://www.facebook.com/fabguru2015>
https://vimeo.com/user5178713 <https://vimeo.com/user5178713>
http://fabgurublog.tumblr.com <http://fabgurublog.tumblr.com/>

Jaknil, 2018-06-11:
I have tried replacing all the glide blocks on humphrey 2 from solid plastic with dry lubricant [Ertalyte ® TX](http://www.boedeker.com/ertaly_p.htm?utm_source=google&utm_medium=cpc&utm_campaign=Ertalyte&utm_content=ertalyte&utm_term=ertalyte&gclid=Cj0KCQjwu_jYBRD8ARIsAC3EGCJTTdM0VmiQMfy559bWI8msx-GFd42NzXiWPIiUJrVDcyjbf4ZEix4aArsJEALw_wcB) to [UHMW PE tape](https://www.amazon.co.uk/gp/product/B00DE2RUMC/) on aluminium glide blocks. Still sliding on untreated aluminium rails. This stopped the pinching beaviour I had before and made tuning way easier. However, black aluminium oxide powder from the rails still quickly builds up on the glide block sliding surfaces, making them "useless" for sliding. We measured the friction force after a few days use to more than double compared to a clean glide surface.

I am VERY intrerested in seeing a picture of the glide block surface on Humphrey 1 for comparison. @uomou (Alex) can you supply this? 

Here are the [Ertalyte ® TX](http://www.boedeker.com/ertaly_p.htm?utm_source=google&utm_medium=cpc&utm_campaign=Ertalyte&utm_content=ertalyte&utm_term=ertalyte&gclid=Cj0KCQjwu_jYBRD8ARIsAC3EGCJTTdM0VmiQMfy559bWI8msx-GFd42NzXiWPIiUJrVDcyjbf4ZEix4aArsJEALw_wcB) Y-rail-glide blocks I took of the machine. The black stuff is alu-oxide. You can clearly see that they were pinching at the screws. At this point the material under the oxide layer does not matter, it is sliding only on alu-versus-alu causing massive friction forces.
[image]

JensDyvik, 2018-06-13:
Here are some sketches for how Humphrey2 could be converted to roll on 608 (skateboard type) bearings. The system is very similar to [Jake Read's](https://gitlab.cba.mit.edu/jakeread/rtcmill), but with classic eccentric bushings instead of flexing structural parts.

- The bearings should roll on UHWN tape ([Jon Nordby pioneered the use of UHMW tape on fabricatable machines](https://github.com/jonnor/tapecore/tree/master/README.md#tapexy)). This will prevent wear and denting of the aluminium, give some much needed dampening, reduce noise, and make the axes repairable on site with hand tools.
- The Y rails can be spaced down with a 16mm Valchromat strip that will allow bearings to roll on the top surface of the rail (right now there is only a 8mm gap for the glide block.
- The tight spot between the gantry feet on the underside of the X axis can be solved by using one bearing in the center instead two on either side
-Eccentric bushings can easily be cnc milled
- Only the axes plates on the gantry feet and gantry will need to be replaced (marked in green on the screenshots)
-Covers for the bearings can be 3D printed or CNC milled from thick plastic stock

A test axis is currently in fabrication by @Jaknil . More news soon!

 If tests are successful this may be an interesting future upgrade for Humphrey1 in Amsterdam if needed.

[image]
[image]
[image]
[image]
[image]
[image]
[image]
[image]

Jaknil, 2018-06-15:
First tests of roller axis are promising, here are [videos](https://www.youtube.com/watch?v=YntWIE5nQJg&list=PLKSI6XgWmHYbdBKBy4OpFnty4vu3g5wap)
After 2000 repetitions we see that this system also generates plenty of oxide in the high load areas, however if vipers are used to push it away and keep dust out it looks promising. We tried adding sawdust to the tracks and it quickly resulted in pocking of the surface so dust protection is key.

Quick data: 
+/-0.03 tolerance with 33,8 kg load on the carriage, load was offset to the side.
Moves incredibly much easier compared to glide blocks, even without oxide on glide blocks.

Jaknil, 2018-07-03:
I have now added my test to the documentation, see the [wiki](https://github.com/fellesverkstedet/fabricatable-machines/wiki/Roller-axis)

jakeread, 2018-07-09:
sorry to switch links on you - that documentation is now [here](https://gitlab.cba.mit.edu/jakeread/rctgantries) 

I like the approach, the key is to be able to tune the bearing preload and I see you are already doing that with the eccentric mounting - that's great! not sure I have much more to contribute here, looking forward to chatting at FAB14.

JensDyvik, 2018-07-09:
Thank you for the updated link Jake. The roller coaster machine looks awesome! I like that we agreeing on the relevance of simple 608 bearing based axes. So far the crazy high part count seems to be outweighed by a lot of advantages *(especially low demands for CNC machining skills to fab parts and low demands for bearing quality)*.

It looks like Torchmate and Precision Plasma LLC has had commercial machine kits based on 608 bearings for years, which should be good indicator of the robustness of a high quantity of low quality bearings.

We are also exploring inline skate wheels as an alternative to smaller v-wheels. Video of the first test [here](https://youtu.be/7Ns-z3jZd3Q).

The link Jakob posted on roller bearing docu has been moved [here](https://github.com/fellesverkstedet/fabricatable-machines/wiki/Fabricatable-axis#roller-rail). The wiki is getting some love and reorganizing these days.

Jaknil, 2018-07-26:
Since the roller axis solution seems to have solved the high friction issue I am closing this issue for now. Big thanks to everyone who contributed!

closed 2018-07-26 by Jaknil