fellesverkstedet/fabricatable-machines #18: PCB patch Outline Does not work correctly.

theholyduck, 2018-09-09:
As per or previous conversations i decided to try the patch suggested in https://github.com/fellesverkstedet/fabricatable-machines/blob/master/humphrey-large-format-cnc/humphrey_v3/img/electronics/possibel_upgrade.JPG

Everything still seems to work but there is no power-out detection at all. Since the drawing there is of a slightly different revision of the pcb than the one we actually have. i have included this drawing of where i cut and where i patched. 
[image]

theholyduck, 2018-09-09:
actually i did just find something that is affected by that patch. with that patch enabled i cant fire up the spindle at all. with it disabled. i can

theholyduck, 2018-09-09:
i guess this patch isnt strictly speaking necesary. the arduino in the original config does seem to pretty reliably detect a power-cut.

Jaknil, 2018-09-12:
I have found why it didn't work. It was two things, the signal was not inverting correctly with that setup and I had mixed up the GND (for arduino) with GNDS for the regulator. I have updated the schematic and board accordingly.  (HRBL-nano). 

Since your machine is already correctly throwing an error message when external power is cycled I suggest that you leave it as it is. 

This is the tested and working schematic to constantly send an ABORT on loss of exernal power:
[image]

closed 2018-09-12 by Jaknil