fellesverkstedet/fabricatable-machines #20: hblr review

jonnor, 2018-09-28:
Suggested changes, in order of priority

* Avoid 5V and 5V_ISO next to Arduino inputs. Possible explanation for PROBE noise
* Avoid 36V planes into/over signal layers
* Add 10uF on 5V, parallel with exiting 10nF
* Add 10uF on 5V_ISO, parallel with exiting 10nF
* Add resistors for stronger pullups to Arduino inputs

CC @Jaknil

jonnor, 2018-09-28:
Should also check current consumption on 5V_ISO, and consider going to a switchmode supply if high.

Jaknil, 2018-10-10:
Updates added to latest version:
- Moved probe lines away from 5V lines
- Moved 36V planes away from optocouplers.
- Added 10uF on 5V, parallel with exiting 100nF
- Added 10uF on VIN, parallel with exiting 100nF (correct?) 
- Added 1k resistors for stronger pullups to Arduino inputs- 
- Added PWM out capacitor pads for smoothing of signal.

JensDyvik, 2018-10-10:
Super. Does this mean that the hrbl shield is now robust enough to be forked as a shopbot controller replacement? I remember we had noisy probe signal on Hedy PCB mill when the shopbot was running next to it.

Jaknil, 2018-10-10:
Yes, that is my opinion. 

If we in the future find that we need another capacitor or similar it's easy enough to add to the forked design. We could of course tweak and slim it indefinitely, to optimize component choices etc but I think it's good enough for forking! 

Regarding the noise on the probe: The stronger pull up on the arduino inputs should make it much more noise resistant and I have made the other suggestions that @jonnor suggested as well.

JensDyvik, 2018-10-15:
In the Hedy version power in and polarity is not on the silk screen. Is this included in the gerber version for production?

Is the Ardunio Nano name on the silkscreen? *(not critical, but will help beginners assemble the card)*

Is the fabricatable Fabricatable machines logo still on? https://github.com/fellesverkstedet/fabricatable-machines/blob/master/wiki-media/fabricatable-machines-logo-single-stroke.dxf

Jaknil, 2018-10-15:
I'll add the logo! 

Last time it was winged on the laser machine due to bad focus. 
This is the front silk-screen as of now:
[image]

Jaknil, 2019-12-09:
+ and - is inverted compared to the plugs on the motors = higher risk of misstakes. Consider swapping.

Jaknil, 2019-12-10:
Adding 3 pin limit switch plugs on the board frees up another pin to be used as signal GND for alarm, making the plug wireing much simpler compared to splitting the GND yourself at the motor. And it is more flexible than having special motor PCBs that split the signals there. If we are doing motor cards then they should do more.

Jaknil, 2020-01-06:
**The factory soldered version of HRBL has a critical error,** the 5V voltage regulator pinout is all wrong. I was to quick when changing to a new regulator that the supplier had in stock.

Datasheet for UTC 78D05AL that comes with the card:

[image]

Board view (seen from the component side of the card):

[image]

Jaknil, 2020-01-06:
**Hot fix for the problem:**

1. Cut the legs of the already soldered regulator or remove it entirely if you can.
2. Solder a new regulator to the pads.

Details:
Here I use a L7805CV 5V regulator fra [Kjell o co link](https://www.kjell.com/no/produkter/elektro-og-verktoy/elektronikk/halvledere/spenningsregulatorer/luxorparts-spenningsregulator-7805-5-v-p90485) (I have 5 in stock) preferably add a [cooling fin](https://www.kjell.com/no/produkter/elektro-og-verktoy/elektronikk/electromechanics/kjoligere-peltier/kjoleribbe-to220-p90351)
but that's probably not critical.

The pinout of the loose L7805CV:
[image]

This is what we want to do:
[image]

Bend the legs and solder directly to the old pads like this

[image]

[image]

And add the cooling fin to make sure it does not overheat. The card acted like a cooling fin previously,

Alternatively, solder the new regulator into these vias and solder a wire to the unused middle pad.
[image]

OBS! Before plugging it in, make double sure that the legs of the factory soldered regulator is no longer connected.

Jaknil, 2020-01-06:
Notes to next version:
- Logo got lost on silk-screnn (too thin lines?)
- Add version nr to the silk screen
- Replace the 5V regulator or change the pins, possibly go for switch mode with 50V tolerance
- Add double y axis singnal ouputs (needs new pin config, what to drop?)
- Add 3 new 3 pin plugs on the board for the limit switches, use the lim pin as a extra signal gnd on the existing plug
- Move the DC power pins to the end of the connector so that it is easier to use another cable thickness for them.

Jaknil, 2020-11-16:
Todo: 
Test the new serial header, does it work as planned?
Test the disable-auto reset cap, if it does not remove the "reset while in motion"-problem, test more options like modding the Arduino board.
Update the boardhouse-files to use the correct linear regulator.