fellesverkstedet/fabricatable-machines #21: Faster electronics wiring with next gen HRBL shield

JensDyvik, 2018-10-15:
Our [HRBL shield](https://github.com/fellesverkstedet/fabricatable-machines/blob/master/hrbl-shield/README.md) has helped speed up the process of wiring motors and control boards. But i think there are financial and accessibility benefits from making it even faster. Our current luxury socket plugs are costly and the 0.5mm2 8pin ribbon cable is not easy to source.

I propose a shift to a common [IDC 16pin 1.27mm pitch 0.127mm2 cable](https://no.farnell.com/pro-power/r2651dtsy16sc85/ribbon-cable-16-core-28awg-30/dp/2628351) type with common female [16pin 2.54mm pitch IDC connectors](https://no.farnell.com/amphenol/t812116a101ceu/socket-idc-s-relief-2-54mm-16way/dp/2215250?st=idc%20socket). I also propose the following pinout *(this pinout will not short if the cable or connector is flipped during assembly)*:

[image]

As we move towards providing ready made [motor shields for integrated motors](https://github.com/fellesverkstedet/fabricatable-machines/blob/master/hertz-axis-monitor/README.md) in combination HRBL shields we can have IDC connectors on both sides, bringing us quite close to plug and play motion control.

### Advantages of current connectors and cables:
- Clamping together IDC connectors is faster than individual wire stripping and more beginner friendly
- Smaller footprint on board
- Cheaper components
- Easier to source components
- Easier and cheaper to order PCB-A's *(ready made boards with soldered components from factory)*
- Thinner wires will be easier to connect to JMC integrated motor connectors
- The spare pin can used as an orientation check *(a led could light up when properly connected)* or spare

### Disadvantages:
- A little less rugged
- When a connector is not connected and the board is live the exposed male pins are easy to short
- 16pin 1.27mm pitch cables are inconsistent in cross section. Some are 1.27mm2 and others 0.072mm2
- If a motor shield is not used on the JMC integrated steppers 16 wires needs to be wirestripped

Jaknil, 2018-10-16:
Interesting suggestion! I really like the idea of using the IDC to save time connecting things and I agree that the price of the connectors we use today is annoyingly high.

**Some wire calculations:**
Your link says  IDC 16pin 1.27mm pitch **0.127mm2** cable, this is not correct. The 0,127mm is the diameter of each copper strand in each wire. The copper cross section diameter of the linked ribbon cable is 0,081mm^2 (correct for AWG28). 

I find that using the AWG values is the best way to get the actual mm^2 copper in the wire. Just a reminder that a smaller AWG = thicker cable.)

I see that the total copper area when using 4 of AWG28 (0,081mm^2) wires for each polarity is tot 0,32 mm^2, this can be compared to the AWG20 (rainbow ribbon) we used before, it has 0,52mm^2/wire. So even using 8 of the 16 wires for power delivery gets us only to 63% of the copper area we had with the rainbow cables. 

Funnily enough it might not matter so much. The stepper driver chops the voltage until it gets the current it wants through the motor coils, so it will not "notice" the resistance in the wires. As long as the wires don't heat up too much or we choose a too small power supply it might be just fine using those wires. Regarding wire heating, they power wires are on the sides, that's good, and it's 3W of heat in total over a 1 meter ribbon and that sounds like very little. So in conclusion I think it will work. But we will loose ca 12W in heat to all the ribbons. It is unconventional, which might lead to people not trusting it, but it should work and would be a nice all in one package.

**Other research**:
I also checked the 16pin connector and it should be able to take the current as well.

I tried to research a 10 pin IDC connector that was large enough to take this 20kr/m AWG20 10 pin ribbon cable https://no.farnell.com/pro-power/pp001508/ribbon-cable-10core-20awg-300v/dp/2776633
But I could not find such a connector, for that cable I think we are best of with the ones we have used before. And I agree that these ribbon AWG20 cables should be considered exotic.

However, I would like to raise an alternative that we can compare your suggestion to. Call it the conventional way with a separate power cable and a signal ribbon:

**Power / Signal separate**
8pin Signal AWG28 Ribbon and IDC connector (costs similar to the 16pin version)
 https://no.farnell.com/amphenol/t812108a100ceu/socket-idc-2-54mm-8way/dp/2215231
https://no.farnell.com/3m/3365-08-100/ribbon-cable-8-way-30-5m/dp/2064465
[image]

2pin Power AWG18-26 Cable and IDC connector (cost very little and we need them for the PSU to board connection anyway)
https://no.farnell.com/amp-te-connectivity/3-640426-2/housing-18awg-2way/dp/1580081
https://no.farnell.com/amp-te-connectivity/640445-2/header-vertical-0-156-2way/dp/589044?MER=sy-me-pd-mi-acce
https://no.farnell.com/jsc-wire-cable/2440-0100s-b-r/no-of-conductors-2conductor/dp/2836435
[image]
[image]
[image]

**Comparison**

Price-wise they are similar and inexpensive:
(Circa prices)

What | price | unit | Quantity | Tot
-- | -- | -- | -- | --
16 core | 8,97 | kr/M | 4 | 35,86666667
16 pin connector | 4,5 | kr st | 8 | 36
TOT 16Pin |   |   |   | **71,86666667 kr**
8 pin connect | 3 | kr st | 4 | 12
8 core wire | 7,4 | kr/M | 4 | 29,6
2 pin connector | 1,48 | st | 8 | 11,84
2 core wire | 2,51 | kr/m | 4 | 10,04
TOT 8+2pin |   |   |   | **63,48 kr**

The main difference in the end is that the user needs to route either a 16 wire cable and 2 connectors or a 8 and 2 wire with total 4 connectors which I feel is kind of similar. Otherwise I think they both fall under the same pros and cons as above.

When used with plain integrated motors the 8+2pin alternative will make it nearly impossible to mix up signal and power and is much easier to connect compared to bunching up 4 wires into the screw connector. That makes me want to go for that option for the cards we order now, and then reevaluate later when we have more progress on motor mounted shields.

JensDyvik, 2018-10-17:
I agree that 8+2pin is the better option for now. Fast and safe wiring on the driver side + safer power lines (larger cross section ) are good arguments.

The cnc milling version of the board could maybe have power and signal connectors on opposing sides of the board. This will reduce the amount of vias needing manual fabrication. 

On a side note, it could be interesting to include Tx and Rx pins in the wire harness in the future. Octoprint has recently gotten support for action commands: http://docs.octoprint.org/en/master/features/action_commands.html

With futere shields for intergrated steppers we could highjack the serial communication to send messages that will stop the machine and trigger popup dialogues like: "Heavy vibration on X axis. Please check milling bit for wear, your settings or fixturing before continuing" or "X motor is not able to move freely, click here for troubleshooting help"

closed 2018-10-17 by JensDyvik

Jaknil, 2018-10-17:
Review with Jon: 
* General comment: An even more traditional approach is to split off power distribution to it's own card and not mix power and signal on one board. The power distribution board would then typically have fuses and flipped-polarity protection. 
* Consider that the 2 pin power connector referenced above is easy to insert backwards by mistake.