fellesverkstedet/fabricatable-machines #25: Problems with squaring and tuning y-axis

newtoneas, 2019-04-04:
We find it hard to tune the Y-axis of our Humphrey. X-axis and Z-axis keeps a good tuning, but y-axis keeps getting out of tuning. To have the machine running precisely, we have to have the motors really tight, which leads to a distorted surface when cutting alongside y. We have a couple of ideas for why it doesn't let itself get properly tuned: 

1. The two rails are not completely aligned, so that when you run the gantry on the y-axis it gets pushed out of tuning. We have no good method for getting the rails completely aligned, when mounted on a valchromat bed. 

2. There might be small dings in the surface on the rails, which might affect the precision. However, the dings on the x-axis is as large, and is not affecting performance. 

Do anyone have any other ideas? We are also trying to exchange out the ball bearings with new ones to see if that improves stuff. When Jakob tuned the machine first time, it was better, however, quite soon it got kicked out of good tuning.

newtoneas, 2019-04-05:
So. Exchanged all outer ball bearings. It did help a bit, but not nearly enough. Have made some measurements on how parallell the y axis are. They are not really parallell, and they are also not really straight. I don't really know how to overcome this. Any ideas on how to make them really parallell?

Jaknil, 2019-04-05:
I belive the solution is to align the rails. 

They are adjustable sideways and rotate arount the point where they are linked. 

Here is the guide for how to align the rails:
https://github.com/fellesverkstedet/fabricatable-machines/blob/master/humphrey-large-format-cnc/tuning.md#aligning-the-y-rails---the-first-rail

Jaknil, 2019-04-26:
Did it resolve the issue?

newtoneas, 2019-04-29:
Well, it's better, but not good enough, and the rails are not completely aligned. We are in the process in making a tool for aligning rails, basically a large T, where the bottom of the T has a measurement device in .01 mm margin. This is because; when you first place the gantry in center, you cannot be sure that you tighten the ball bearings in a straight manner, therefore replicating an angle along the rail.

Jaknil, 2019-08-07:
**New firmware feature!**: The latest GRBL build supports auto squaring of the gantry! 
"This optional dual axis feature is primarily for the homing cycle to locate two sides of 
  a dual-motor gantry independently, i.e. self-squaring." [See GRBL Commit](https://github.com/gnea/grbl/commit/b75e5571eeaeb22a88304716fb1e7411f9c28be0) Thanks to @theholyduck from Dalamakers for bringing this to our attention. I know @JensDyvik has been cooking up his own ideas on this before.

Please note that to implement this you need to change the wiring since the pinout changes. Regardless I think it's a worthwhile thing to implement! 

A potential challenge I could see with implementing this is that it takes away the spindle enable pin (D13). The obvious fix is to make the spindle to be always enabled and pull down the PWM pin to reduce the risk of it getting started by noise. Removing pin D13 as spindle enable is actually beneficial since it is the source of the notorious "spindle twitch" as the arduino boots and self-tests.