fellesverkstedet/fabricatable-machines #26: Aluminium milling feeds and speeds - updated values?

Jaknil, 2019-04-26:
I would like to add the new high speed feeds and speeds to the [alu-milling guide](https://github.com/fellesverkstedet/fabricatable-machines/blob/master/humphrey-large-format-cnc/alu_parts.md#how-to-mill-aluminium-parts)

These are the "safe" values I always used so far: 
- 6mm Endmill Up-Cut single-flute: Feed rate 12mm/s, plunge rate 8mm/s, Spindle RPM 18000, pass depth 2.5mm, ramp angle 12deg
- 4mm Endmill Up-Cut single-flute: ? _(I'll try to add these later)_ 
- 3mm Endmill Up-Cut single-flute: Feed rate 9mm/s, plunge rate 6mm/s, Spindle RPM 18000, pass depth 1.2mm, ramp angle 12deg

For higher speed rough slotting Jens mentioned that 3x feed and 1.8mm pass depth for the 6mm bit gave good results if you then used the safe speed for the finishing pass (0.1 mm stock to leave).

Were those the final numbers?

newtoneas, 2019-04-29:
I just want to note that this is very much a matter of the type aluminum used. I've had to use quite different settings when we machined aluminum last time, due to a softer type.

JensDyvik, 2019-05-02:
Yes. New values for roughing is 36mm/s and 1.8mm pass depth. Finishing should be done at a lower feedrate. The old 12mm/s is safe. It is probably possible finish mill at a faster feedrate. But finishing at the same 36mm/s seemed to be too fast and left an uneven edge.

0.1mm stock to leave is a little bit too little for a 6mm bit. 0.18mm or 0.2mm is good. My default is 3% of tool diameter stock to leave

Newtoneas, what type of aluminum alloy did you mill and which settings? We have typicically used 6082 aluminium.

Jaknil, 2019-05-26:
[Updated the settings in the humhprey build guide](https://github.com/fellesverkstedet/fabricatable-machines/blob/master/humphrey-large-format-cnc/alu_parts.md#how-to-mill-aluminium-parts)

JensDyvik, 2019-05-28:
Awesome! Closing issue

closed 2019-05-28 by JensDyvik