fellesverkstedet/fabricatable-machines #27: Debriefing notes Humphrey v5

Jaknil, 2019-06-01:
- Dust skirt assembly and principles are not documented (files and parts are documented.) Also suggested redesign to use only one thickness of material (less confusion easier to make separate)
- Slukk sketches i CAM model files to reduce confusion
- Fastener name cyl = cylinder
- Top assembly fusion 3D model is broken
- Electronics guide

nikolai-kolstad, 2019-06-01:
**All**
Rydd i sketches alle modeller
Planlegge holddowns. Kjøre to runder, en alle hull uten og så en alle hull med washers. Ta med 6+mm holddown-hull I dette. 
Rett opp view kontra XYZ 
Origo i toppen av materiale (vær konsekvent)
0.2 ekstra for finishing pass kan være litt lite
Toleranse testbiter i alle deler, også i tre
Anbefalt rekkefølge på fresing

**Gantry**
Jigg for nærme kanten til å feste støtte i x-akse
Rart XYZ-punkt i setup 
Si noe om gjenging i Fusion, ikke nødt
Hvis ikke gjenging, grunnere hull
Doble hull i to scrap parts
Frese bort stor del i safety cleanup
Contour marking og cut har forskjellige path

**Y-rails**
Spre delene for bedre slotting og mindre scrap
I tooth pattern left går den til et startpunkt mellom hver rekke

**Y rail holder**
4 mm bit I jiggoppsett
Holddown 2 + 3 overlapp?
Når generere toolpaths holddown 2+3 genereres masse ekstra hull
4mm pockets genererer empty toolpath

**x-z rail**
Mulig å sette holddown i hull i pocket som senere freses (frese alt m 4mm?)
Cutof uten rampe?
Problemer m chamfer (lage test part?)

**Sheet 2 8mm**
Dfx open 3mm slots

**Sheet 1 8mm**
Mangler doku på pocket dybde

**Wiki**
Tools mangler bor
Bits mangler 4mm

Jaknil, 2019-06-01:
Dust skirt source models fusion 360, not clean: https://a360.co/2HpvjlT

Dust skirt assembly now documented here: https://github.com/fellesverkstedet/fabricatable-machines/blob/master/humphrey-large-format-cnc/Carriage_assembly.md#z-axis-spindle-and-dust-skirt

Jaknil, 2019-06-01:
Priority Jakob:
- Top assembly fusion 3D model is broken
- Toleranse testbiter i alle deler, også i tre
- Anbefalt rekkefølge på fresing til instruks
- Origo i toppen av materiale (vær konsekvent)
- Si noe om gjenging i Fusion, ikke nødt
- Doble hull i scrap parts (make sure, which ones are missing?)

Annoying but no easy fix:
- View origin vs CAM setup origin.
- tooth pattern left går den til et startpunkt mellom hver rekke

Priority Nikolai:
- Sheet 2 8mm, Dfx open 3mm slots
- Sheet 1 8mm, Mangler doku på pocket dybde