fellesverkstedet/fabricatable-machines #32: Wooden humphrey documentation?

Jaknil, 2019-11-21:
I wish there was more documentation about the making of the wooden humphreys. 

[Humphrey wood version - how to make](https://github.com/fellesverkstedet/fabricatable-machines/blob/master/humphrey-large-format-cnc/wood_version_how_to_make.md) is all computer graphics, I know several are built or in the process, don't we have any pictures of how it looks? 

I am looking into how to make a new wood-based CNC and this would be nice to have.

hwalseng, 2019-11-21:
What is the most interesting to know or get pictures of? I can upload the pictures I have from the process of making it, gluing it and onwards, one of the coming days.

Jaknil, 2019-11-21:
Sounds like great stuff to upload! I bet it will be very appreciated, a pic with a short description goes a very long way.

I advise agains waiting “til later” with documentation. I don’t believe that ever happens. Upload while it’s fresh and edit later = much more accessible and relevant :)

hwalseng, 2019-11-21:
I added a link to a google drive folder with pictures and video from my build, to the wooden Humphrey readme. It's not a complete log (far from that). But I hope the pictures and video can be of some help, and I'll provide som commentary (or just rename the files) someday soon (hopefully :-P )

https://github.com/fellesverkstedet/fabricatable-machines/blob/master/humphrey-large-format-cnc/wood_version_how_to_make.md

Jaknil, 2019-11-25:
Thanks! I added a few pictures to the readme:s, looks lots better already! 
* https://github.com/fellesverkstedet/fabricatable-machines/blob/master/humphrey-large-format-cnc/README.md
* https://github.com/fellesverkstedet/fabricatable-machines/blob/master/humphrey-large-format-cnc/README.md#how-to-build-a-wood-gantry-humphrey

The commenting of the pictures will be really nice. Here are two versions of how it was done for earlier humphreys:
* https://github.com/fellesverkstedet/fabricatable-machines/blob/master/humphrey-large-format-cnc/Humphrey_v4.md
* https://github.com/fellesverkstedet/fabricatable-machines/blob/master/humphrey-large-format-cnc/humphrey_v2.md

closed 2019-11-25 by Jaknil

JensDyvik, 2020-05-04:
Added lots of pictures, descriptions and design wins and fails: https://github.com/fellesverkstedet/fabricatable-machines/blob/master/humphrey-large-format-cnc/wood_version_how_to_make.md

Still work to be done, but we are progressing :)