fellesverkstedet/fabricatable-machines #33: Spindle control with BLDC-DZZ and HRBL card

Jaknil, 2019-11-25:
[image]

There was a question on how to make this (brushless) spindle driver work with the HRBL card.

Jaknil, 2019-11-25:
[The manual](http://www.aero-mate.com/images/Windward/BLDC-DZZ%20User%20manual.pdf) 

First I would run their startup and test guide to see that it works and has the correct nr of poles set.
Please document this testing since it isn't very clear in the manual.

When you have tested it and want to control it with the HRBL card
These are the connections that need to be made:
 [image]

The switches on the top right corner also needs setting.
You want the external controller mode, not the CNC mode (since we are not sending it digital signals). See chapter: 8.2 Mode B. Manual mode2- operating from an
External control 

Switches:
[8] CNC
[7] External
[6] Speed-ADJ RP
[5] CNC-ADJ PWN
[4] Open loop (for max power always, I am not sure about this one, test it)
[3] CW (Direction of spin, flip if it is wrong)
[2] NTC OFF (it's the motor temperature sensor, unless you have one, then it can be ON!)
[1] Stop (should not matter, since we set it to listen for CNC on [8]) 

Configuring GRBL:
Set max RPM in firmware settings in [UGS ](https://winder.github.io/ugs_website/)or type:
$30 = your-maximum-rpm-value
I'll use 60000 rpm in this example. 
$30 = 60000
And also set the miniumum rpm value to 0 to avoid scaling issues.
$31 = 0
[Offical guide to configuring GRBL spindle speed](https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#30---max-spindle-speed-rpm)

Testing the spindle with GRBL:
Type into the console of [UGS](https://winder.github.io/ugs_website/)
Start the spindle at 60000 rpm
M3 S60000
Stop the spindle
M5
Just set the current RPM to 20000
S20000

Troubleshooting:
If it doesn't work, try putting switches:
[1] to Start 
or 
[6] to CNC
and try again. Document which one that worked.

Let me know if there are more questions and PLEASE write down a guide as you go through this, write it straight into github or it is likely to not happen.

hwalseng, 2019-11-26:
@nikolai-kolstad if we manage to do the wiring together at OsloMet this or next week, it would be great to get some help documenting the procedure with pictures and comments, for other makers to benefit from.

hwalseng, 2019-12-09:
Added a folder with a README and img folder.
https://github.com/fellesverkstedet/fabricatable-machines/tree/master/Spindle%20setup/250W%2040000rpm
Still a work in progress, but I have added more information and pictures.
Main troubleshooting going forward: switches position, get the correct rpm, and find out about [menu] and [poles] button on spindle driver.

Jaknil, 2019-12-14:
Regarding the [DIP switches positions ](https://github.com/fellesverkstedet/fabricatable-machines/tree/master/Spindle%20setup/250W%2040000rpm#switch-flip-on-spindle-driver) on the spindle driver. 

I want to start with some general advice. Test one thing at a time and in as small steps as possible. 

**Incorrect speed control**
Tip: Make sure you set the minimum speed in the GRBL firmware settings in UGS to 0 rpm and the max to the max of the spindle, anything else messes up the signal. If that is correct then I think the problem is in the driver configuration, number of poles etc. But if you want to verify the card: Measure the voltage at the input to the driver by probing the SPEED ADJ and GND inputs. It should read 5V = max speed and 2.5V = half speed etc... If the voltage is correct when you compare it to the set speeds in UGS then the problem is in the driver settings.  

**Spindle runs slowly when forced on**
You mention that it always runs slowly  when you manually turn it on with switch 1. I would say that that is just as expected. We want it turn it on and off with the control card signal from GRBL.

To fix this my first test would be: Does the spindle start when you through UGS send GRBL the start spindle command and a speed, like: "M5 S20000"? (M5 is start and M3 is spindle stop)
If you need to flip switch 1 to START to make it run, then the driver does not listen to our digital start command. A possible fix is to change the setting on switch 8 which controls which start signal it should listen to, I think up is correct, you mention yours is down. 

To verify the card: You could measure that the signal actually changes on the inputs marked "switch" on the driver by measuring the voltage between them when GRBL gets a M5 or M3 command (start or stop spindle).