fellesverkstedet/fabricatable-machines #34: Soft reset, HRBL PCB problem?

hwalseng, 2019-12-10:
I got a 'soft reset' alarm in UGS and i couldn't move the machine. The unlock button in UGS didn't solve anything. Neither did a reset of the arduino, or turning the power for the whole machine off and on.

hwalseng, 2019-12-10:
Wiggling the HRBL card solved the problem. Probably samebug as in this video:
https://drive.google.com/file/d/1Pn09c_3s9rNyox6e-3luZ04rpS2rWmSm/view?usp=drivesdk
(GRBL throws soft resett alarm when the HRBL shield is tightened to much with screws to a structure, in my case)

Jaknil, 2019-12-18:
I think the most likely culprit is that there is something wrong with the solder joints somewhere on the card. I suggest you swap it for a machine soldered card as soon as we get the connectors to avoid risks of intermittent electrical errors in the future. Impressive that you found how to trigger it!