fellesverkstedet/fabricatable-machines #37: Problems with the herring axis alu_extrusion_herring_rack

uli6666, 2022-12-14:
Hallo,
I love your alu_extrusion_herring_rack design, but it won´t work for me.
If I opened it everything seems to be ok, but if a changed the length in the spreadsheet from 120mm to maybe 200mm I get the following errors:

00:45:04  Traceback (most recent call last):
  File "C:\Users\win10\AppData\Roaming\FreeCAD\Mod\freecad.gears\freecad\gears\features.py", line 106, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "C:\Users\win10\AppData\Roaming\FreeCAD\Mod\freecad.gears\freecad\gears\features.py", line 567, in generate_gear_shape
    c = obj.clearance
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'clearance'
00:45:04  Traceback (most recent call last):
  File "C:\Users\win10\AppData\Roaming\FreeCAD\Mod\freecad.gears\freecad\gears\features.py", line 106, in execute
    gear_shape = self.generate_gear_shape(fp)
  File "C:\Users\win10\AppData\Roaming\FreeCAD\Mod\freecad.gears\freecad\gears\features.py", line 234, in generate_gear_shape
    fp.da = "{}mm".format(fp.gear.da)
<class 'AttributeError'>: 'FeaturePython' object has no attribute 'da'

I do know nothing about Freecad, please can you help me out.
Thank you in advance

JensDyvik, 2023-01-04:
Hi

Have you installed the "Gears" workbench in Freecad? Its an external workbench that can be installed from -Tools - Addon manager.

I have not yet tested out this herring rack for alu extrusions design, so I dont know yet if it is a good idea or not. Please report if you have any success!

uli6666, 2023-01-04:
Hi,
yes it´s installed. If I started it everything is O.K., but if I changed the length the errors that I described above comes on. Maybe a older version of "Gears"??
There is no newer version for Rhino or Solidworks, because I like the design that is self centered in fits the nut.
Thank you
Ulrich

JensDyvik, 2023-01-04:
I tested now and get the same bug as you. I think the problem is that the newer version of the gears workbench does not work well the file from 2020. The only option is to rebuild the file. So for now the file unfortunately is not parametric and only works as a reference. It is however possible to change the number of teeth on the pinion. 

I did test to manually copy and paste the existing rack, and the merge at the last hole. That gave a 214mm long rack. I have uploaded the file for you here: https://github.com/fellesverkstedet/fabricatable-machines/blob/master/Module%20development/herring-axis/alu_extrusion_herring_rack_214mm.FCStd

Would that work for you?

[image]

closed 2023-01-12 by JensDyvik